sdr-enthusiasts/docker-tar1090 #4: tar1090 location does not map to readsb container

rucknapucknavitz, 2020-05-05:
The mikenye/docker-tar1090 location does not correctly map to the lat/lon set in mikenye/docker-readsb

The default location is over Manhattan, NY.
Pressing the H (H)ome/Reset Map key on the browser will reset the map on NY.

@wiedehopf suspects that the tar1090 container has its own readsb, if so it will need its own lat lon set as well

mikenye, 2020-05-06:
Hi @rucknapucknavitz, this has been solved. The docker image `mikenye/tar1090:latest` and `mikenye/tar1090:commit_630eed9` contain the fix. Please pull one of these images and test. Thanks.

rucknapucknavitz, 2020-05-06:
It's still not working for me. I pulled the :latest (docker-compose pull and then docker-compose up -d) and that didn't work. I then updated my docker-compose.yml with LAT and LONG per the update to the guide (https://github.com/mikenye/docker-readsb/wiki/Guide-to-ADS-B-Data-Receiving,-Decoding-and-Sharing,-Leveraging-RTLSDR-and-Docker#deploying-tar1090-for-improved-visualisation) and that also didn't work. I'm still stuck in NY, unfortunately. Location on the map in PiAware and PlaneFinder containers work fine.

rucknapucknavitz, 2020-05-06:
Bingo. Re-pulled following the latest update here and it's working great! Map defaults to my location. Now shows distance rings, and distance from antenna in the data columns on the right. Thank you!

closed 2020-05-06 by rucknapucknavitz