sdr-enthusiasts/docker-tar1090 #8: PR: Allow Range Outline

bacco007, 2020-06-21:
The base tool allows for you to include a range outline - https://github.com/wiedehopf/tar1090#heywhatsthatcom-range-outline

It would be good to be able to allow a map to a local file to enable this functionality

mikenye, 2020-06-22:
Hi @bacco007,

Thanks for the suggestion.

Rather than map to a local file, I've instead implemented a new container environment variable `HEYWHATSTHAT_PANORAMA_ID`.

If you follow the steps to create a panorama, then pass the panorama ID through to `HEYWHATSTHAT_PANORAMA_ID`, the panorama JSON will be downloaded during container start, and be shown on the map.

Pull the latest version of the image and give this a try, and let me know how you go.

bacco007, 2020-06-22:
Added the variable, but the docker logs report

```[nginx] 192.168.1.74 - - [22/Jun/2020:16:38:35 +1000] "GET /upintheair.json HTTP/1.1" 404 136 "http://192.168.1.64:8078/" "Mozilla/5.0 (Windows NT 10.0; rv:77.0) Gecko/20100101 Firefox/77.0"```

mikenye, 2020-06-22:
Can you post the logs from the container startup?

Up to the line `[services.d] done.`.

Thanks.

bacco007, 2020-06-22:
don't mind me - I deployed my updated docker-compose file to the wrong server :)

It works great - thanks!

mikenye, 2020-06-22:
Awesome!

closed 2020-06-22 by mikenye