sdr-enthusiasts/docker-tar1090 #27: /heatmap is empty

Olivier6767, 2020-10-07:
Hi, First of all thanks for your great work!
I've setup **readsb** connected to my RTLSDR receiver and the following containers are pulling form it: **tar1090**, **graph1090**, **piaware**, **rbfeeder**, **fr24feed**, **adsbx**, **opensky**

Everything works fine except that when accessing `http://dockerhost:8078/?heatmap` or `http://dockerhost:8078/?heatmap&realHeat` like mentioned in the readme, I get an empty map...

The folder `/var/globe_history` does contain data for each day (ttf files).

Am'I missing something ?

Olivier6767, 2020-10-07:
It seems that the issue is a permissions problem. In the docker logs, I have:
`[nginx] 2020/10/07 11:43:13 [error] 30857#30857: *22 open() "/var/globe_history/2020-10-07/heatmap/03.bin.ttf" failed (13: Permission denied), client: 10.1.1.75, server: _, request: "GET /globe_history/2020-10-07/heatmap/03.bin.ttf HTTP/1.1", host: "tar1090.redacted.com", referrer: "https://tar1090.redacted.com/?heatmap"`

My docker file contains:
```
    volumes:
      - /mnt/NAS-Docker/data/node-sdr/tar1090/history:/var/globe_history
```

I've tried to set owner to **www-data:www-data** and permission to **0777** on the /globe_history folders but I still get the permission error in the logs.

Am'I missing something?

mikenye, 2020-10-07:
Hi @Olivier6767,

You're very welcome, annd I'm sorry you're having problems.

Try the following.

1. Get a shell inside the container, eg: `docker exec -it tar1090 bash`
2. Reset the ownership to `root:root` on the `/var/globe_history` directory & contents: `chown -vR root:root /var/globe_history`
3. Reset the permissions on the `/var/globe_history` directory & subdirs: `find /var/globe_history -type d -exec chmod -v 0755 {} \;`
4. Reset the permissions on the files under `/var/globe_history`: `find /var/globe_history -type f -exec chmod -v 0644 {} \;`

Let me know if this helps.

Olivier6767, 2020-10-07:
Thanks for the information. With this I was able to find out that my volume mount to my NAS (/mnt/NAS-Docker) forces permissions to be 0777 for all files and folders. When running your commands in the post above, it indicates the permissions are changed but when looking back, they are still the same.
It seems related to the way I export my shared folders (no_root_squash). I'm trying to figure this out and will report here once I found a solution. 
Thanks!

mikenye, 2020-10-07:
No problems, glad you're on the right track! :-)

Olivier6767, 2020-10-07:
I was able to fix my permissions. The heatmap (and realHeat) are not fully functional.
Thanks again for your great work!

closed 2020-10-07 by Olivier6767

mikenye, 2020-10-07:
Awesome! I’m glad you got it sorted. :-)