sdr-enthusiasts/docker-tar1090 #28: Feature Request: Add Timelapse functionality

Olivier6767, 2020-10-08:
Would it be possible to include the **Timelapse** functionality from **wiedehopf/timelapse1090** either in the `docker-piaware` (uses dump1090-fa), or better, in the `docker-readsb` or `docker-tar1090` containers?

This would be a nice addition.

Thanks again for the great work!

mikenye, 2020-10-09:
I've implemented this..

Pull the latest version of the container and set container environment variable `ENABLE_TIMELAPSE1090=true` to enable.

Let me know if there are any issues, I've not used timelapse1090 before...

Olivier6767, 2020-10-09:
I pulled the latest version of the container but it does not include the new code:
- no `timelapse1090` in `/etc/services.d`
- no  `location /timelapse` section in etc`/nginx/sites-enabled/tar1090`

It looks like the build does not include the updated code...
I tried to pull `mikenye/tar1090:latest` both on **arm32v7** and **amd64** platforms.

Olivier6767, 2020-10-10:
I pulled the image again today and the code is now included.
The `timelapse` feature works now perfectly!

Thanks for the quick turnaround and awesome support!

closed 2020-10-10 by Olivier6767