sdr-enthusiasts/docker-tar1090 #31: Aircraft database parse error

speedst3r, 2020-11-16:
Possibly related to https://github.com/wiedehopf/tar1090/issues/77 , the aircraft database js files are gzipped and not being parsed correctly in the client, leaving the aircraft type blank and resulting in browser dev console errors:

```
7c1c6f: Database load error: parsererror at URL: db2/7.js              planeObject.js:2023 
76bd49: Database load error: parsererror at URL: db2/7.js             planeObject.js:2023 
7c7181: Database load error: parsererror at URL: db2/7.js              planeObject.js:2023 
```

I am able to manually decompress the js files and rename them which solves the issue, it looks like nginx is not returning the ```Content-Encoding: "gzip"``` header.

```
$ curl -I http://192.168.0.216:8078/db2/7.js
HTTP/1.1 200 OK
Server: nginx
Date: Mon, 16 Nov 2020 10:31:01 GMT
Content-Type: application/javascript
Content-Length: 147661
Last-Modified: Mon, 16 Nov 2020 10:29:22 GMT
Connection: keep-alive
ETag: "5fb25482-240cd"
Cache-Control: public, max-age=7776000
Accept-Ranges: bytes
```

I changed the location stanza in /usr/local/share/tar1090/nginx.conf from
```
  location ~ db-.*\.js$ {
    gzip off;
    add_header Cache-Control "public, max-age=7776000";
    add_header Content-Encoding "gzip";
  }
```
to
```
  location ~ db2/.*\.js$ {
    gzip off;
    add_header Cache-Control "public, max-age=7776000";
    add_header Content-Encoding "gzip";
  }
```
and it seems to be working correctly:
```
$ curl -I http://192.168.0.216:8078/db2/7.js
HTTP/1.1 200 OK
Server: nginx
Date: Mon, 16 Nov 2020 10:33:08 GMT
Content-Type: application/javascript
Content-Length: 147661
Last-Modified: Mon, 16 Nov 2020 10:29:22 GMT
Connection: keep-alive
ETag: "5fb25482-240cd"
Cache-Control: public, max-age=7776000
Content-Encoding: gzip
Accept-Ranges: bytes
```

mikenye, 2020-11-16:
You, sir, are an absolute legend. I’ve been trying to get to the bottom of this for a few days now.

mikenye, 2020-11-16:
Did you want to submit a PR or do you want me to fix it up?

speedst3r, 2020-11-16:
I can take a stab at it later if you're not in a hurry, but I'm thinking it would be this command somewhere in the Dockerfile:

```sed -ie "s/db-/db2\//" /usr/local/share/tar1090/nginx.conf```

(it's what I ran with docker exec on the live container)

mikenye, 2020-11-16:
I ended up putting the fix in place, and new images are currently building. They should be ready in about 30 mins or so. Thanks again for pinpointing the issue. It is appreciated! I've credited you in the PR.

closed 2020-11-16 by mikenye

speedst3r, 2020-11-17:
Yep, that's probably a better fix than what I was thinking! Thanks!