sdr-enthusiasts/docker-tar1090 #45: add environment variable to enable FA links

katchy3132, 2021-04-26:
please add an environment variable to enable FlightAware links 
(the first sed expression worked for me ) 

https://github.com/wiedehopf/tar1090#enable-disable-fa-links-in-the-webinterface-previously-enabled-by-default

>  #  ENABLE:
> sudo sed -i -e 's?.*flightawareLinks.*?flightawareLinks = true;?' /usr/local/share/tar1090/html/config.js
> #  ENABLE if the above doesn't work (updated from previous version)
> echo 'flightawareLinks = true;' | sudo tee -a /usr/local/share/tar1090/html/config.js
> #  DISABLE:
> sudo sed -i -e 's?.*flightawareLinks.*?flightawareLinks = false;?' /usr/local/share/tar1090/html/config.js

mikenye, 2021-05-21:
PR #50 adds `TAR1090_FLIGHTAWARELINKS`, which you can set to any value to enable the FA links in the web interface. Sorry for the delay on this one!

mikenye, 2021-05-21:
The updated image will be live on docker hub once #150 completes

commit 8ebd6b28 (2021-05-21): Add TAR1090_FLIGHTAWARELINKS for #45 (#50)

Closes #45
closed 2021-05-21 by mikenye