sdr-enthusiasts/docker-tar1090 #54: Feature Request: make range ring colors configurable, too

jstammi, 2021-06-15:
tar1090 allows to configure the color of each range ring defined by `SiteCirclesDistances` by way of similar array named `SiteCirclesColors`. If less colors are givem in the array than range rings are defined, the last color used for the remaining.

E.g. by appending
`SiteCirclesColors = new Array('#ffffff');`
to
`/usr/local/share/tar1090/html/config.js`
inside your running tar1090 container all range rings become white colored.

The change seems trivial: use another environment variable, e.g. `TAR1090_RANGERINGSCOLORS`, in
`rootfs/etc/cont-init.d/04-tar1090-configure`
similar to the already existing `TAR1090_RANGERINGSDISTANCES`. Only pitfall may be the need of single quotes around the colors. Needs mentioning in documentation or addition while handling the `$TAR1090_RANGERINGSCOLORS`.

mikenye, 2021-06-15:
Good suggestion. I will implement this.

mikenye, 2021-06-16:
Thanks to @fredclausen for implementing.

closed 2021-06-16 by mikenye

jstammi, 2021-06-24:
Thanks a lot for the effort, I can confirm it is working :-)