sdr-enthusiasts/docker-tar1090 #58: READSB_EXTRA_ARGS seems to be ignored

sgofferj, 2021-08-20:
I have set "READSB_EXTRA_ARGS=--json-trace-interval=1" but it doesn't seem to be honored. When I exec into the container I see the variable set to the correct value but ps shows me
```
/usr/local/bin/readsb --net-only --quiet --lat 61.255098 --lon 24.056373 --net-connector=adsbx_feeder,30105,beast_in --net-connector=adsbx_readsb,30005,beast_in -
```

sgofferj, 2021-08-20:
THanks to some midnight debugging aid from Wiedehopf I found that the variable is actually honored but --json-trace-interval=15 seems to be hardcoded somewhere:
```
root@f6db290f66f8:/# pgrep -a readsb
540 /usr/local/bin/readsb --net-only --quiet --lat 61.255098 --lon 24.056373 --net-connector=adsbx_feeder,30105,beast_in --net-connector=adsbx_readsb,30005,beast_in --write-json=/run/readsb --heatmap-dir /var/globe_history --heatmap 13 --json-trace-interval 15 --json-reliable 1 --max-range=300 --net-ro-port=30002 --net-sbs-port=30003 --net-bi-port=30004,30104 --net-bo-port=30005 --net-json-port=30047 --forward-mlat --json-trace-interval 0.1
```

sgofferj, 2021-08-21:
@wiedehopf found out that the reason for my issues was in his code and the parameter hardcoding was not the root cause of it. The parameter hardcoding itself remains but doesn't appear to cause problems. Feel free to close or fix "for beauty".

wiedehopf, 2021-08-21:
That's just a container default for the heatmap the container has.
The extra args will correctly take precedence over the earlier arguments and thus overwrite the default.

closed 2021-08-23 by mikenye