sdr-enthusiasts/docker-tar1090 #76: timelapse1090 data not persisting

trenchcoatjedi, 2022-01-30:
Running the latest container. timelapse1090 data does not persist but heatmap data does.

In the docker-compose.yml I have the following:

volumes:
  timelapse1090:
  tar1090_heatmap:

Under the tar1090 container I have the environment variable set ENABLE_TIMELAPSE1090=true and under volumes:

    volumes:
      - timelapse1090:/var/timelapse1090
      - tar1090_heatmap:/var/globe_history

mikenye, 2023-02-09:
Hi @trenchcoatjedi,

This is the behaviour of timelapse1090 itself.

When starting up, it performs some `rm` commands: https://github.com/wiedehopf/timelapse1090/blob/8eb023b878a2e80e5a22e4eb81bcc1194c08940d/timelapse1090.sh#L34

I'm not sure what the impact of disabling these `rm` commands would be.

I would suggest opening an issue on the wiedehopf/timelapse1090 repo and asking the question.

-Mike

kx1t, 2023-03-01:
Cleaning up -- is it OK to close this issue?
It looks like the solution is not with the container, but with the underlying code and user was advised to open an issue there.

Thanks --kx1t

closed 2023-03-02 by mikenye