sdr-enthusiasts/docker-tar1090 #90: ENABLE_978=yes then got 978.json: No such file or directory

DesertOnSea, 2022-03-17:
Hi, great works!

When adding `ENABLE_978=yes` to the environment options, I got this in docker logs:

`tar1090       | [tar1090] /usr/local/share/tar1090/tar1090.sh: line 104: 978.json: No such file or directory`

Then I tried these the error goes away, but the default `URL_978` (at http://127.0.0.1/skyaware978) keeps returning 404 Not Found:

```
docker exec -it tar1090 bash
touch /run/readsb/978.json
```

Am I missing anything? Thanks!

chevyman142000, 2023-01-20:
I am seeing the same error. Was there any resolution to this?

mikenye, 2023-01-23:
This one is difficult for me to troubleshoot, as we don't have UAT in Australia (where I am located). I'll look into this but would be happy to accept a PR if anyone finds the solution.

kx1t, 2023-02-08:
In order to make this work, you need to have a working SkyAware978 web instance deployed. This means that you should deploy (for example) the [docker-dump978](https://github.com/sdr-enthusiasts/docker-dump978) container*, or otherwise have this software deployed.

Then you need to set 2 parameters:
```yaml
- ENABLE_978=yes
- URL_978=http://dump978/skyaware978
```

The `URL_978` should be the URL at which your SkyAware978 webpage is available -- if `docker-dump978` runs in the same container stack, then the above line should work. If you run `dump978` elsewhere, then you should figure out what the correct URL is.

Alternatively, you can also pull in UAT data by ingesting it with a `--net-connector` entry, see the documentation for the format.

* Note that we made an update to the [docker-dump978](https://github.com/sdr-enthusiasts/docker-dump978) container today that actually exposes the skyaware978 website -- please do a `docker pull` of this container and recreate/restart it!

closed 2023-02-09 by mikenye