sdr-enthusiasts/docker-tar1090 #109: feature request: add influx and Prometheus data so it’s compatible with what `docker-readsb-protobuf` provides

kx1t, 2023-03-02:
For Prometheus, I already mapped http://xxx/metrics -> /run/readsb/stats.prom but the data available there is different and much less than what’s generated by docker-readsb-protobuf. Ideally, the fields and values should be compatible with what docker-readsb-protobuf provides so the grahams dashboards can be reused. 

Thanks!

dirkhh, 2023-03-02:
Definitely interested to see this. Happy to help if I can.

kx1t, 2023-03-25:
hey @mikenye , with the appearance of our [ultrafeeder](https://github.com/sdr-enthusiasts/docker-adsb-ultrafeeder) that combines tar1090 with the multifeeder's multiple MLAT clients and integrates an MLATHUB, it would be great if we could get this working so tar1090 (and ultrafeeder) would be able to integrate with Grafana dashboards.

I'd do it myself but I have very little knowledge about how to create this data stream. --thanks!

PR #113 (2023-03-26): Add Telegraf
Adds the ability to use Telegraf to provide InfluxDBv2 and Prometheus compatible metrics.
Closes #109.
commit dcdfa5f8 (2023-03-26): Add Telegraf (#113)

Adds the ability to use Telegraf to provide InfluxDBv2 and Prometheus
compatible metrics.
Closes #109.
closed 2023-03-26 by kx1t