sdr-enthusiasts/docker-tar1090 #110: `timelapse1090` should write to `/run/timelapse1090` instead of to `/var/timelapse1090`

kx1t, 2023-03-21:
It does a LOT of disk writes, and even if you keep the files around, they will not persist after a restart of `readsb` according to @wiedehopf. 

He advises to write to  `/run/timelapse1090` instead of to `/var/timelapse1090`

[[tracking issue so we don't forget about it]]

mikenye, 2023-03-22:
> He advises to write to `/run/timelapse1090` instead of to `/var/timelapse1090`

This is because `/run` should be `tmpfs` (in memory).

mikenye, 2023-03-22:
Looking at: https://github.com/sdr-enthusiasts/docker-tar1090/blob/02a85bb9c89566a89f0f0fe7b7f45b7b7a4d42df/rootfs/etc/services.d/timelapse1090/run#L6

I can't remember why I did this... Possibly so the data was persistent.

kx1t, 2023-03-25:
So I am going to close this request. After further deliberation, we want users NOT to use the timelapse1090 functionality, but the included (newer) http://ipadress/?replay functionality instead. That one is using data that is already stored in persistent storage and barely adds any reads/writes to the SD unless someone is actively browsing to the replay web page.

I've updated the documentation to indicate that this is the way to go.

Also, it appears that timelapse1090 is disabled by default, so a user would have to enable it actively to use it. In that case, we can assume that they know what they are doing.

Thanks for looking at this, Mike :)

closed 2023-03-25 by kx1t