sdr-enthusiasts/docker-tar1090 #136: Support for RTL_TCP

mscottco, 2023-07-09:
Hi Team,
I currently use this container with an RTL-SDR dongle but I'd like to migrate to RTL_TCP so the SDR can sit in a more convenient location. Is it possible to configure/add support for RTL_TCP as a data source?
Cheers!

kx1t, 2023-07-10:
@wiedehopf - does your readsb support rtl_tcp input, and if so, how’s that configured?

kx1t, 2024-02-04:
It doesn't look like we support this.

However, you can set up a remote device with an SDR and run either [Ultrafeeder in a minimalized form](https://github.com/sdr-enthusiasts/docker-adsb-ultrafeeder?tab=readme-ov-file#minimalist-setup), or install readsb on the bare metal. From there, you can feed your data to a local tar1090 or ultrafeeder instance that can be used for display and for sending this data to other aggregators.

Good luck - if you need any further help, please join the Discord server linked in the README.md document

closed 2024-02-04 by kx1t