sdr-enthusiasts/docker-tar1090 #249: Page doesn't refresh with frp share.

faifaint, 2024-12-13:
I try to build a mikenye/tar1090 docker container at my local server(Ubuntu 24.04),
 feed by some local readsb hardware, 
and share my local 8078 port with frp to my remote web server, 
now I can get my web tar1090, 
but this page doesn't refresh.

**docker-compose configuration:**
version: "3.8"
services:
  tar1090:
    image: ghcr.io/sdr-enthusiasts/docker-tar1090:latest
    container_name: tar1090
    restart: always
    environment:
      - TZ=Asia/Shanghai
      - BEASTHOST=readsb
      - LAT=43.915068
      - LONG=87.559737
      - S6_SERVICES_GRACETIME=30000
      - UPDATE_TAR1090_DAYS=15
      - PTRACKS=24
      - TAR1090_ENABLE_AC_DB=true
      - READSB_ENABLE_TRACES=true
      - TAR1090_USEROUTEAPI=true
      - READSB_NET_CONNECTOR=192.168.50.42,30002,raw_in;192.168.192.8,30002,raw_in;192.168.50.111,30002,raw_in
      - READSB_EXTRA_ARGS=--write-json-globe-index --write-globe-history /var/globe_history
    volumes:
      - /opt/adsb/tar1090/globe_history:/var/globe_history
      - /opt/adsb/tar1090/graphs1090:/var/lib/collectd
      - /proc/diskstats:/proc/diskstats:ro
      - /usr/local/share/osm_tiles_offline:/usr/local/share/osm_tiles_offline
    ports:
      - 8078:80
    tmpfs:
      - /run:exec,size=64M
      - /var/log
 
**frpc configuration:**
serverAddr = "xx.xx.xx.xx"
serverPort = 7000
auth.token = "**********"

[[proxies]]
name = "tarweb2"
type = "http"
localPort = 8078
customDomains = ["tar2.xxx.xxx"]

**frps configuration:**
bindPort = 7000
webServer.addr = "0.0.0.0"
vhostHTTPPort = 8080
auth.token = "**********"

wiedehopf, 2024-12-13:
can you try describing better what you mean with "doesn't refresh"
by far the easiest would be to just link the page to be honest.
this might be too small for the extra args you selected:
`- /run:exec,size=64M`
with that config i'd also recommend enabling:
`- READSB_ENABLE_API=true`
it should work without as well though
does it work without the proxy?

faifaint, 2024-12-13:
It means http://tar2.faifaint.top only shows map,but the aircraft doesn't move, and messages number does not change.
Container runs fine on local ip 192.168.50.5:8078

faifaint, 2024-12-13:
Docker logs:
[2024-12-13 15:57:26.814][readsb] [2024-12-13 15:57:26.814 CST] readsb starting up.
[2024-12-13 15:57:26.814][readsb] readsb version: 3.14.1645 wiedehopf git: c3214b3 (committed: Tue Nov 19 16:39:37 2024 0100)
[2024-12-13 15:57:26.814][readsb] Using lat: 43.915068, lon: 87.559737
[2024-12-13 15:57:26.820][readsb] loading state .....
[2024-12-13 15:57:26.828][readsb] actual range outline, read bytes: 368644
[2024-12-13 15:57:26.829][readsb]  .......... done, loaded 140 aircraft in 0.009 seconds!
[2024-12-13 15:57:26.829][readsb] aircraft table fill: 0.0
[2024-12-13 15:57:26.829][readsb] 30002: Raw TCP output port
[2024-12-13 15:57:26.829][readsb] 30005: Beast TCP output port
[2024-12-13 15:57:26.829][readsb] 30006: BeastReduce TCP output port
[2024-12-13 15:57:26.830][readsb] 30047: Position json output port
[2024-12-13 15:57:26.830][readsb] 30003: SBS TCP output ALL port
[2024-12-13 15:57:26.830][readsb] 32006: SBS TCP input MAIN port
[2024-12-13 15:57:26.830][readsb] /run/readsb/api.sock: API output port
[2024-12-13 15:57:26.830][readsb]
[2024-12-13 15:57:26.830][readsb] 32007: SBS TCP input MLAT port
[2024-12-13 15:57:26.830][readsb] 32008: SBS TCP input PRIO port
[2024-12-13 15:57:26.831][readsb] 32009: SBS TCP input JAERO port
[2024-12-13 15:57:26.831][readsb] /run/readsb/cmd.sock: command input port
[2024-12-13 15:57:26.831][readsb] 30001: Raw TCP input port
[2024-12-13 15:57:26.832][readsb] 30004: Beast TCP input port
[2024-12-13 15:57:26.832][readsb] 30104: Beast TCP input port
[2024-12-13 15:57:26.838][graphs1090] Generating all graphs
[2024-12-13 15:57:26.853][readsb] Beast TCP input: Name resolution for readsb failed: Temporary failure in name resolution
[2024-12-13 15:57:26.854][readsb] Raw TCP input: Connection established: 192.168.50.42 port 30002
[2024-12-13 15:57:26.854][readsb] Raw TCP input: Connection established: 192.168.50.111 port 30002
[2024-12-13 15:57:26.966][readsb] Raw TCP input: Connection established: 192.168.192.8 port 30002
[2024-12-13 15:57:27.436][readsb] Beast TCP input: Name resolution for readsb failed: Temporary failure in name resolution
[2024-12-13 15:57:27.608][readsb] Database update in progress!
[2024-12-13 15:57:27.681][readsb] Database update first part took 0.252 seconds!
[2024-12-13 15:57:27.841][readsb] Database update done! (critical part took 0.001 seconds)
[2024-12-13 15:57:28.724][readsb] Beast TCP input: Name resolution for readsb failed: Temporary failure in name resolution
[2024-12-13 15:57:28.829][tar1090-update] curl: (28) SSL connection timeout
[2024-12-13 15:57:28.830][tar1090-update] [ERROR] Error in line 72 when executing: DB_VERSION_NEW=$(curl --connect-timeout 2 --silent --show-error "https://raw.githubusercontent.com/wiedehopf/tar1090-db/master/version")
[ERROR] /etc/s6-overlay/scripts/tar1090-update in line 17 when executing: "${s6wrap[@]}" bash /tar1090-install.sh /run/readsb webroot "${TAR1090_INSTALL_DIR}"
cat: /var/globe_history/tar1090-update/git/version: No such file or directory
[ERROR] /etc/s6-overlay/scripts/tar1090-update in line 18 when executing: TAR1090_VERSION=$(cat "${TAR1090_UPDATE_DIR}/git/version")
[2024-12-13 15:57:30.842][readsb] Wrote live traces for aircraft active within the last 15 minutes. This took 4.0 seconds (roughly 0 minutes).
[2024-12-13 15:57:31.379][readsb] Beast TCP input: Name resolution for readsb failed: Temporary failure in name resolution
[2024-12-13 15:57:31.811][tar1090] 2969 (process ID) old priority 0, new priority 19
[2024-12-13 15:57:34.843][readsb] Wrote live traces for aircraft active within the last 24 hours. This took 4.0 seconds (roughly 0 minutes).
[2024-12-13 15:57:35.591][readsb] Beast TCP input: Name resolution for readsb failed: Temporary failure in name resolution
[2024-12-13 15:57:35.806][collectd] [2024-12-13 15:57:36] plugin_load: plugin "logfile" successfully loaded.
[2024-12-13 15:57:38.774][graphs1090] Done with initial graph generation
[2024-12-13 15:57:43.548][readsb] Beast TCP input: Name resolution for readsb failed: Temporary failure in name resolution
[2024-12-13 15:57:58.993][readsb] Beast TCP input: Name resolution for readsb failed: Temporary failure in name resolution
[2024-12-13 15:58:29.418][readsb] Beast TCP input: Name resolution for readsb failed: Temporary failure in name resolution
[2024-12-13 15:59:00.309][readsb] Beast TCP input: Name resolution for readsb failed: Temporary failure in name resolution
[2024-12-13 15:59:31.204][readsb] Beast TCP input: Name resolution for readsb failed: Temporary failure in name resolution

wiedehopf, 2024-12-13:
Your proxy (or CF) is setting cache flags.
Your proxy (or CF) is being naughty.

```
https://tar2.faifaint.top/data/globe_0020.binCraft.zst
cache-control: public, max-age=14400
```
The container is responding for the data/globe files with:
```
Cache-Control: public, no-cache
```

If the browser caches the data, you're not gonna get updates.
So fix your CF settings.

Also remove this, you probably don't want it:
```
- READSB_EXTRA_ARGS=--write-json-globe-index --write-globe-history /var/globe_history
```
READSB_TRACES probably already does what you want ... not sure what exactly you want ofc.

If you leave json-globe-index on, set
```
- READSB_ENABLE_API=true
```
That will make the updating nicer.

faifaint, 2024-12-13:
Yes, I set my CF pass cache ,It works ,Thanks for your patience.

wiedehopf, 2024-12-13:
I'd ask you to send data to adsb.lol or airplanes.live or adsb.fi, but it's probably not allowed.

But would you please close the issue, thank you!

Have a good day!
And if you want to discuss antennas / reception, the SDR enthusiast discord is probably a good place.

faifaint, 2024-12-13:
Yes, the country I am in does not allow sharing such data abroad. Thank you very much for your assistance.

closed 2024-12-13 by faifaint

wiedehopf, 2024-12-13:
[image]

[image]

You don't need to bypass the cache entirely, this is the required setting.