sdr-enthusiasts/docker-tar1090 #271: ?replay not working on Chrome version 134.0.6998.89

domeus, 2025-03-12:
?replay not working on chrome latest version, map is empty. Working fine with latest firefox.

wiedehopf, 2025-03-12:
Not adding any more workarounds for the chromium browser bug.

Wait for this to be in chrome: https://issues.chromium.org/issues/401652934

wiedehopf, 2025-03-12:
also replay works for me.

wiedehopf, 2025-03-12:
I guess it's the typical behaviour of this bug that it's intermittent.
At least if there is enough planes on the map.

wiedehopf, 2025-03-12:
This browser bug is so annoying.

Nonetheless, added a workaround.
Restart the container to get the tar1090 update.

This can be closed.

domeus, 2025-03-12:
> This browser bug is so annoying.
> 
> Nonetheless, added a workaround. Restart the container to get the tar1090 update.
> 
> This can be closed.

Wonderful, now working! Thank you 👍 :)

closed 2025-04-25 by fredclausen